daid/LegacyCura #1555: Gcode generation in "TYPE SKIN"

irmas, 2017-02-02:
My custom printer is having issues printing SKIN types. 

My printer (basic and custom made) only prints when there is an E command on the line. In theory the Gcode I posted below fills my part 100%.  In practice, it is is only partly filled (because 2/3 of the lines do not contain an E command). 

* Why isn't there an E command on all of the lines ? 

* Is there a way to force Cura to generate an E command at the end of each line ?

`;TYPE:SKIN
G1 X256.424 Y149.737 E228.26498
G0 X256.565 Y149.596
G0 X256.308 Y148.722
G1 X248.724 Y156.305 E229.04434
G0 X248.583 Y156.446
G0 X247.838 Y156.061
G1 X256.061 Y147.838 E229.88943
G0 X256.202 Y147.697
G0 X255.716 Y147.051
G1 X247.05 Y155.717 E230.78004
G0 X246.909 Y155.858
G0 X246.344 Y155.291

etc`

BagelOrb, 2017-02-03:
Could you please attach your full gcode?

What version of Cura are you running?

If you're running Cura 2.4, could you attach your workspace?

Ghostkeeper, 2017-02-06:
You could make a post-processing script (called "plug-in" in legacy Cura) that does this for you. Every G1 command gets preceded by your "Turn extrusion on" command. Every G0 command gets preceded by your "Turn extrusion off" command.

venuasg, 2019-12-25:
Hi,

Can you let me know from where the below line settings is picked from cura
G1 X256.424 Y149.737 E228.26498

Thanks