daid/LegacyCura #1556: Maker Select 3D/Cura 3.21.  "The printer is not connected"

TTVert, 2017-02-17:
win10 64Bit, selected a Prusi i3 but also tried to create a custom printer to no avail, Works fine on Cura 15.0.4.6 and I confirmed I can communicate w/ COM3 at 115200.  It seems others have this same issue and I'm not sure where to look. Looking at the log files in  USBPrinting don't even mention it looking to any port.  It appears this is all auto configured now but I cannot find a log indicating what it is trying to do or where it is failing.  

I've attached the two log files I'm aware of that may have something to do with this but any other help would be appreciated.

Thanks
Dave

Ghostkeeper, 2017-02-17:
Your attempt to attach log files failed. At least, I don't see any.

What version of Cura are you using? We haven't released any `Cura 3.21`.

BagelOrb, 2017-02-20:
We at Ultimaker don't really support 3rd party printers.
We accept pull requests though ;)

TTVert, 2017-02-20:
Fair enough, but why list printers as options to choose that clearly don't work?

Dave

TTVert, 2017-02-20:
Sorry Ghostkeeper, I had to zip them as log files weren't allowed. I've attached.

Ghostkeeper, 2017-02-23:
We don't have those 3rd party printers, so we can't test whether they still work other than trying to slice with them just in the interface. Stuff like the printer not connecting is impossible to test without buying the actual printer.

TTVert, 2017-02-23:
If I could fix them I'd design the software myself.  In this case the software is auto detecting hardware however it isn't even trying to auto detect usb devices per the log.  All I need is the ability to manually specify the COM port and baud rate w/in your software.  Your 15.x software allows this and the printer prints fine.

BagelOrb, 2017-02-24:
This issue is logged under the repo of Cura 1.15.04.

You should log your issue under github.com/Ultimaker/Cura so that it actually arrives at the Cura team which is currently developing Cura 2.

Ghostkeeper, 2017-02-24:
It was also logged under the Cura repository, BagelOrb: https://github.com/Ultimaker/Cura/issues/1434

TTVert, 2017-02-24:
Yeah sorry I had it in the wrong place, fairly new to github.  So far no luck figuring anything out.