daid/LegacyCura #1558: cura layers not printing

gazzab99, 2017-03-06:
Hi, 
my model in Cura 14.07, had 914 layers.
I saved it to gcode to go into my Anet A8 via SD card.
the print stopped early, and when checking the gcode file back in Cura it only had 595 layers.
I am very new (a few days) to 3d printing, so Im hoping there is something in a setting somewhere that I've missed. any help would be appreciated.

BagelOrb, 2017-03-06:
Why are you running such an ancient version of cura? Cura 1.14 is from
2014. Cura 2 has been out for a year. Cura 2.4 is just out and cura 2.5
beta is soon to be released.

Give it a try. We're not developing on cura 1 any more.

Ghostkeeper, 2017-03-07:
You probably used a different layer height back then, or scaled the model differently.

914 is about 50% more than 595 (actually 53.6%).