daid/LegacyCura #1559: Cura not able to use Mac USB full size keyboard's numeric key pad

StephenInoue, 2017-03-08:
Noticed that when I go to scale a model or any other action where I would enter a numeric value in cura that you are unable to see the numbers I'm typing. 

Can you find a USB wired keyboard with numerica pad and verify that the Mac Cura client is able to use the numeric keypad?

This would be a really nice feature to have and I'm not sure why Cura isn't supporting it.

BagelOrb, 2017-03-08:
Try Cura 2. Cura 1.15 is not being developed any more....

StephenInoue, 2017-03-09:
BagelOrb - Thanks! The old Cura's check for update said I was up to date, but I totally missed version 2. Thanks!

closed 2017-03-09 by StephenInoue

BagelOrb, 2017-03-10:
@daid People are still on the old Cura because they don't know about a new one. Isn't it easy to give the mthe update to download the new Cura?

daid, 2017-03-10:
@BagelOrb when we did that we where suddently faced with terabytes of data transfers from china...

BagelOrb, 2017-03-10:
@daid WTF?!

So... is that a good or a bad thing?
Doesn't that mean that a lot of users in China are using Cura?
Or is there something shifty going on?

daid, 2017-03-10:
@BagelOrb Don't know the details, ask Jasper.