daid/LegacyCura #1562: 10thousandHours!

KyleMaker, 2017-03-25:
I have no clue why this is happening. When i loaded a file that normally takes about 10 min to print it says 1500 hours. Ive tried just about everything but no results yet if anyone has had this problem before i would apreciate some feedback. thnx

BagelOrb, 2017-03-25:
Have you tried Cura 2? It has loads of improvements over cura 1.15

daid, 2017-03-26:
We also looked in our magic all seeing globe, but it provided no additional info on your problem. Our local mind reader and hacker have also been unable to access your mind and computer. So without additional info we are stuck on helping you.

Ghostkeeper, 2017-03-27:
Maybe the speed was set to 0.0001mm/s or the layer height to 1 carbon atom's width.