daid/LegacyCura #1564: Why 0 minute?Cura 15.04

a56919, 2017-03-29:
I don't know how to discribe the ... I can see the model, but the Time shows '0 minutes' or nothing, and Nothing in the Layers Part, only show 1 piece here. It run well yesterday ... can't save the G-code.
[image]

[image]
[image]

Ghostkeeper, 2017-03-30:
It sounds like something went wrong in the slicing process. Are you certain the model fits inside the build volume?

It also looks like something went wrong when uploading those images. We can't see them.

Cura 2 might solve your problems. A lot has been redesigned since 15.04.

a56919, 2017-04-02:
[image]
[image]
thanks, 
It run well a few days ago, the same model. I'll try the Cura 2.